OpenKinect/libfreenect #563: Trouble building python wrappers on Windows with MinGW

Etherian, 2018-08-02:
I am attempting to build and install libfreenect with the Python 3 wrapper on Windows 7 with MinGW64. I was able to build successfully without the Python wrappers (after the hours of debugging that building anything on Windows requires, of course) but when I configure the build to include the Python 3 wrapper, mingw32-make consistently reports the errors below.

build log:
````
[ 15%] Built target freenectstatic
Already have audios.bin
[ 15%] Built target firmware
[ 30%] Built target freenect
[ 33%] Built target freenect_sync
[ 36%] Built target freenect-tiltdemo
[ 40%] Built target freenect-camtest
[ 43%] Built target freenect-wavrecord
[ 46%] Built target freenect-regtest
[ 50%] Built target freenect-glview
[ 53%] Built target freenect-micview
[ 56%] Built target freenect-regview
[ 60%] Built target freenect-hiview
[ 63%] Built target freenect-chunkview
[ 66%] Built target freenect-glpclview
[ 73%] Built target fakenect
[ 78%] Built target fakenect-record
[ 81%] Built target freenect_sync_static
[ 85%] Built target freenect-cppview
[ 88%] Built target freenect-cpp_pcview
[ 91%] Built target freenect_cv
[ 95%] Built target freenect-cvdemo
[ 96%] Linking C shared module python3\freenect.dll
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x438): undefined reference to `__imp__PyThreadState_UncheckedGet'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x45f): undefined reference to `__imp_PyFrame_New'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x8c3): undefined reference to `__imp_PyCode_New'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x23ea): undefined reference to `__imp_PyObject_CallFinalizerFromDealloc'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x2474): undefined reference to `__imp_PyObject_CallFinalizerFromDealloc'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x24be): undefined reference to `__imp__PyType_Lookup'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x2b1b): undefined reference to `__imp_PyUnicode_AsUnicode'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x2c8b): undefined reference to `__imp_PyUnicode_AsUnicode'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x2dc2): undefined reference to `__imp_PyUnicode_AsUnicode'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x3989): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x3a5f): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x3bba): undefined reference to `__imp__PyType_Lookup'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4721): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x48a3): undefined reference to `__imp__PyThreadState_UncheckedGet'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4bc0): undefined reference to `__imp__PyThreadState_UncheckedGet'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4c37): undefined reference to `__imp_PyFrame_New'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4cee): undefined reference to `__imp__PyObject_GetDictPtr'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4d0b): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x4e48): undefined reference to `__imp_PyCode_New'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x528f): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x529c): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x588b): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x5b0d): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x616f): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x617c): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x695b): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x6b3d): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x6ecb): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x7060): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x7073): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x7dc6): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x7f35): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x7f48): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x8d46): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x8e15): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x90f6): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x91c5): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x9a87): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x9b96): more undefined references to `__imp__PyDict_GetItem_KnownHash' follow
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xaab6): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xace7): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xaf17): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xb147): undefined reference to `__imp_PyFunction_Type'
[... 12 lines cut ...]
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xd891): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xda03): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xdade): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe1d6): undefined reference to `__imp__PyThreadState_UncheckedGet'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe282): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe29d): undefined reference to `__imp_PyFunction_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe5d5): undefined reference to `__imp_PyMethod_Type'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe811): undefined reference to `__imp__PyDict_GetItem_KnownHash'
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0xe9d8): undefined reference to `__imp_PyMethod_Type'
[... cut ...]
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text+0x1063b): more undefined references to `__imp__PyDict_GetItem_KnownHash' follow
CMakeFiles\cython3_freenect.dir/objects.a(freenect3.c.obj):freenect3.c:(.text.unlikely+0x3b3): undefined reference to `__imp__PyThreadState_UncheckedGet'
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [wrappers\python\CMakeFiles\cython3_freenect.dir\build.make:106: wrappers/python/python3/freenect.dll] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1039: wrappers/python/CMakeFiles/cython3_freenect.dir/all] Error 2
mingw32-make: *** [Makefile:129: all] Error 2
````

HongTang973, 2022-11-09:
hello, did you figure it out this problem. I am with this problem now

Etherian, 2022-12-26:
I did not solve this problem. I am sorry.