OpenKinect/libfreenect #566: Libfreenect Python Failed to claim camera interface

tarunprakash, 2018-08-25:
I am using kinect v1 on macOS High Sierra with libfreenect for python. I am trying to access the depth camera feed on the kinect with

`array,_ = freenect.sync_get_depth()`

This never used to give me problems before, but now, everytime I run the program, I am greeted with this error:

```
Failed to claim camera interface: LIBUSB_ERROR_ACCESS
Error: Invalid index [0]
Error: Can't open device. 1.) is it plugged in? 2.) Read the README
```

However, if I run `freenect-glview` in terminal, there is no problems, and everything works fine. I have tried replugging the device, but it doesn't seem to work.

seanngpack, 2019-10-13:
Any resolution to this?

MrGiskard, 2024-08-17:
The issue still remains and it's 2024. Any feedback?

Kakcalu13, 2025-01-11:
awww damnit