OpenKinect/libfreenect #569: Failed to access the same kinect v1 in two processes

frannecki, 2018-11-11:
hello,
I'm using kinect v1 for ROS and want to access the camera's video stream at the time, but when i'm trying to access the kinect in another terminal i get the following errors
```shell
Failed to claim camera interface: LIBUSB_ERROR_BUSY
Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.terminate called after throwing an instance of 'std::runtime_error'
  what():  Cannot open Kinect
Aborted (core dumped)
```
And when i'm getting the video the kinect also fails to start in ROS. Has anyone any suggestions? I'm quite new to kinect. Any help would be appreciated

piedar, 2018-11-12:
Normally, you can't have multiple processes access the camera at the same time.  With some care, you could use one process for camera and another for audio/motor.  Or you could try [proxynect](https://github.com/OpenKinect/libfreenect/pull/401).

frannecki, 2018-11-14:
@piedar Thank you. Actually I found out that the camera could be accessed using ROS message with openni. Issue closed.

closed 2018-11-14 by frannecki