OpenKinect/libfreenect #570: Could not open camera LIBUSB_ERROR_ACCESS

Anvolne, 2018-11-25:
I built the library for Windows, all went well, no horrible errors from Visual Studio, but when I tried to run some of the examples it gives a "Could not open camera: LIBUSB_ERROR_ACCESS" error every time, for any of the examples.

Also, the only thing I did is I built it with freeglut instead of the **very** outdated glut, but that shouldn't cause a problem right?

I replaced these drivers with Zadig to libusbK:
Xbox NUI Motor
Xbox NUI Camera
Kinect USB Audio

Actual output from console:
```
D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-glview.exe
Kinect camera test
Number of devices found: 1
Could not open camera: LIBUSB_ERROR_ACCESS
Could not open device

D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-glview.exe
Kinect camera test
Number of devices found: 1
Could not open camera: LIBUSB_ERROR_ACCESS
Could not open device

D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-cvdemo.exe
Could not open camera: LIBUSB_ERROR_ACCESS
Error: Invalid index [0]
Error: Kinect not connected?

D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-cvdemo.exe
Error: Invalid index [0]
Error: Kinect not connected?

D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-camtest.exe
Could not open camera: LIBUSB_ERROR_ACCESS
```

Anvolne, 2018-11-25:
Okay, so after poking around, I resolved the access error
Now it's trying to send commands to it, and failing:
```
D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\wrappers\python>python demo_cv2_async.py
Press ESC in window to stop
send_cmd: Output control transfer failed (-7)
freenect_fetch_reg_pad_info: send_cmd read -7 bytes (expected 8)
freenect_camera_init(): Failed to fetch registration pad info for device
Error: Can't open device. 1.) is it plugged in? 2.) Read the README

D:\Cloud Storage\Google Drive\Szakdoga\libfreenect\bin\Release>freenect-camtest.exe
send_cmd: Output control transfer failed (-7)
freenect_fetch_reg_pad_info: send_cmd read -7 bytes (expected 8)
freenect_camera_init(): Failed to fetch registration pad info for device
Device 000001F23B883DE0 open during shutdown, closing...
```

Anvolne, 2018-11-25:
I have a solution for the latter: It does not work in a USB 3.0/3.1 port... at all. It even works in a USB 2.0 hub too, but not in a 3.0 port...
I can only assume this is a problem with libusb or something

closed 2018-11-25 by Anvolne

hafizbadrie, 2020-04-19:
Hey I'm new to this and I, too, got the same issue. However this was in ubuntu.

I looked at the libusb code here: https://github.com/libusb/libusb/blob/e83f39ba6e0368d8b253a5f0b02bfd115763e0b4/libusb/libusb.h#L1055. This is a permission problem. In debian/ubuntu, I just need to run the command with **sudo** command. 

I'm not sure about windows, though. Perhaps this could lead to something.