OpenKinect/libfreenect #574: libfreenect headers not installed

eriknoorland, 2019-01-09:
Hi,

I'm trying to get started with https://github.com/nguyer/node-kinect on a Raspberry Pi but when i run npm install i get the error **libfreenect/libfreenect.h: No such file or directory**.

So i went back to my install of libfreenect and when i run the command **cmake -L ..** i get the following response:
```
-- Operating system is Linux
-- Got System Processor armv7l
-- libfreenect will be installed to /usr/local
-- Headers will be installed to /usr/local/include/libfreenect
-- Libraries will be installed to /usr/local/lib
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Could NOT find OpenGL (missing:  OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR) 
-- Could NOT find GLUT (missing:  GLUT_glut_LIBRARY GLUT_INCLUDE_DIR) 
-- Could NOT find OpenGL (missing:  OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR) 
-- Could NOT find GLUT (missing:  GLUT_glut_LIBRARY GLUT_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libfreenect/build
-- Cache values
BUILD_AS3_SERVER:BOOL=OFF
BUILD_CPACK_DEB:BOOL=OFF
BUILD_CPACK_RPM:BOOL=OFF
BUILD_CPACK_TGZ:BOOL=OFF
BUILD_CPP:BOOL=ON
BUILD_CV:BOOL=OFF
BUILD_C_SYNC:BOOL=ON
BUILD_EXAMPLES:BOOL=ON
BUILD_FAKENECT:BOOL=ON
BUILD_OPENNI2_DRIVER:BOOL=OFF
BUILD_PYTHON:BOOL=OFF
BUILD_PYTHON2:BOOL=OFF
BUILD_PYTHON3:BOOL=OFF
BUILD_REDIST_PACKAGE:BOOL=ON
CMAKE_BUILD_TYPE:STRING=
CMAKE_INSTALL_PREFIX:PATH=/usr/local
LIBUSB_1_INCLUDE_DIR:PATH=/usr/include/libusb-1.0
LIBUSB_1_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libusb-1.0.so
```

After the **make** command is finished i check the **/usr/local/include** folder and there is no libfreenect folder.

Am i doing something wrong? Any help would be appreciated.

eriknoorland, 2019-01-09:
I fixed it by running **make install**. This is however not mentioned in the build instructions.

closed 2019-01-09 by eriknoorland

WyattAutomation, 2019-07-23:
^ can confirm that does the trick, thanks!