OpenKinect/libfreenect #576: Could not build library after commit d20b594...

iman-keangkasa, 2019-01-30:
Hello. I am using Ubuntu 16.04. I followed installation from source using [these tutorial](https://naman5.wordpress.com/2014/06/24/experimenting-with-kinect-using-opencv-python-and-open-kinect-libfreenect/)

I however stuck when I try to make the library (after CMake -L .. ) and get this error:
`[ 94%] Building CXX object wrappers/cpp/CMakeFiles/freenect-cppview.dir/cppview.cpp.o
In file included from /home/rais/version-control/libfreenect/wrappers/cpp/cppview.cpp:27:0:
/home/rais/version-control/libfreenect/wrappers/cpp/libfreenect.hpp:182:8: error: ‘unique_ptr’ in namespace ‘std’ does not name a template type
   std::unique_ptr<uint8_t[]> m_rgb_buffer;
        ^
/home/rais/version-control/libfreenect/wrappers/cpp/libfreenect.hpp: In member function ‘void Freenect::FreenectDevice::setVideoFormat(freenect_video_format, freenect_resolution)’:
/home/rais/version-control/libfreenect/wrappers/cpp/libfreenect.hpp:115:5: error: ‘m_rgb_buffer’ was not declared in this scope
     m_rgb_buffer.reset(new uint8_t[mode.bytes]);
     ^
wrappers/cpp/CMakeFiles/freenect-cppview.dir/build.make:62: recipe for target 'wrappers/cpp/CMakeFiles/freenect-cppview.dir/cppview.cpp.o' failed
make[2]: *** [wrappers/cpp/CMakeFiles/freenect-cppview.dir/cppview.cpp.o] Error 1
CMakeFiles/Makefile2:810: recipe for target 'wrappers/cpp/CMakeFiles/freenect-cppview.dir/all' failed
make[1]: *** [wrappers/cpp/CMakeFiles/freenect-cppview.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`
Can you help me with this problem. It seems that the error comes from the last commit d20b594.. ( on the cpp wrappers)
My gcc version is:  5.4.0
Thank you so much.

jborj006, 2019-01-31:
I followed the same guide and got those same errors. I used an older version and that helped solve the problem. https://github.com/OpenKinect/libfreenect/tree/debian

iman-keangkasa, 2019-01-31:
Thanks so much @jborj006!. It worked.

closed 2019-01-31 by iman-keangkasa

sumdii, 2019-07-09:
As diogodcarvalho said, add `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}  -std=c++11")` at about 119th line of  CMakeLists.txt in libfreenect root directory,then it worked.