OpenKinect/libfreenect #580: Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO

jefflee0127, 2019-03-16:
please help me i am very urgent 

i am trying to connect my kinect 1517 to ubuntu 18.04 

i downloaded libfreenect and everything, and if i try to execute freenect-glview 
i keep getting this error. 

`freenect-glview
Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO
Found sibling device [same parent]
Could not open device: LIBUSB_ERROR_NO_DEVICE
Could not open device
`
maybe it's because of the usb problem? 
because i don't have any usb bus 003 when i execute lsusb -t 

anyways please help me 
i searched everywhere, but i wasn't able to find the answer... 
please help me..

BrainPotter, 2019-05-06:
I have this problem too. Do you have any answers?

pproctor182, 2019-05-09:
I also have this problem as well: 
```
Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO
Found sibling device [same parent]
Could not open device: LIBUSB_ERROR_IO
```

pproctor182, 2019-05-09:
Mine works after trying this solution from another answer answer in another issue (https://github.com/OpenKinect/libfreenect/issues/543):

Please try freenect-camtest, which uses only the camera.

"The root cause is probably missing audio firmware. Model 1473 uses the audio device to control the motor, and needs firmware for the audio. I am working on a firmware fix, but I still can't get it to work with 1473. In the meantime, I'm thinking of printing out warnings when audio/motor fails to load instead of stopping the program"

jithendrabsy, 2019-05-29:
So, Can't we open Kinect 1473 through OpenCV?
Is there any other way rather than this? (for Model 1473)

yasiupl, 2020-06-12:
What did it for me, was running `freenect-micview` and `freenect-camtest` before `freenect-glview`.

procub3r, 2020-12-31:
> What did it for me, was running `freenect-micview` and `freenect-camtest` before `freenect-glview`.

This worked for me! thanks a lot!

Rage997, 2021-11-28:
> What did it for me, was running `freenect-micview` and `freenect-camtest` before `freenect-glview`.

When I run `freenect-micview` I get:

```
Number of devices found: 1
Trying to open ./audios.bin as firmware...
Trying to open /home/rage/.libfreenect/audios.bin as firmware...
Trying to open /usr/local/share/libfreenect/audios.bin as firmware...
Trying to open /usr/share/libfreenect/audios.bin as firmware...
Trying to open ./../Resources/audios.bin as firmware...
upload_firmware: failed to find firmware file.
upload_firmware failed: -2
Could not open device

```

any help is appreciated.

n-qber, 2021-12-23:
The file `audios.bin` is missing (I'm not sure but I think it is a firmware that needs to be sent to to the kinect during runtime)
To download it, go to the git cloned folder of libfreenect and run:

`python3 ./src/fwfetcher.py`

This will download the audio.bin to the libfreenect folder, then simply move it to the location of the installed libfreenect library

`sudo mv ./audios.bin /usr/local/share/libfreenect/`

NewtonAlbukerk, 2022-09-01:
@n-qber,      

This worked for me! thanks a lot!

Lecrapouille, 2023-03-23:
I have also the model 1473 (I did not know where to get this info: it's depicted in the sticker on the foot of the device) and I had the same error `LED of K4W or 1473 device: LIBUSB_ERROR_IO` with, i.e., freenect-glview . I solved the case by the following two steps depicted in https://github.com/OpenKinect/libfreenect/issues/543:
- what @frederico-klein said: `python3 ./src/fwfetcher.py`
- then what @StoneLabs said: comment the line `fnusb_keep_alive_led(ctx, audio); ` in usb_libusb10.c https://github.com/OpenKinect/libfreenect/blob/aca6046a5bc1bc5bbe7f2ae8c0085464e7ebd9f1/src/usb_libusb10.c#L443

Lecrapouille, 2023-04-22:
Update. On my Ubuntu I got this error again. This is because I was inside the build folder. After doing `sudo make install` and `cd ..` it worked back

AntonyCat, 2024-05-01:
If anyone is still checking this thread... I am attempting to run  `python3 ./src/fwfetcher.py` to download the audio file `audio.bin` but it seems that https://www.xbox.com/system-update-usb is no longer a page on the Xbox website (error 404). 

Would anyone know where I can manually download the files i need to from elsewhere / if someone has the files I can download from them? Much appreciated!!

lucasdaweb95, 2025-06-01:
> If anyone is still checking this thread... I am attempting to run `python3 ./src/fwfetcher.py` to download the audio file `audio.bin` but it seems that https://www.xbox.com/system-update-usb is no longer a page on the Xbox website (error 404).
> 
> Would anyone know where I can manually download the files or update.zip that `fwfetcher.py` used to download from the website? Or if someone has the files I can download from them? Much appreciated!!

i dunno about you but it worked for me.

StyledStrike, 2025-12-14:
While trying to play with this library, I've ran into this issue, and was trying to figure out why `freenect_open_device` succeeds only after running `freenect-micview` once.

As part of that, I was trying to setup the device like how micview does it, and found out something that might be interesting to you guys. If I try to do this:

```cpp
freenect_select_subdevices(fn_ctx, (freenect_device_flags)(FREENECT_DEVICE_MOTOR | FREENECT_DEVICE_AUDIO | FREENECT_DEVICE_CAMERA));
```

Then, calling `freenect_open_device` would fail with the `Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO
` message. However, if I just do this instead:

```cpp
freenect_select_subdevices(fn_ctx, FREENECT_DEVICE_AUDIO);
```

Then, and ONLY then, libfreenect decides to upload the firmware from `audios.bin`, and from now on (until the Kinect is physically disconnected) `freenect_open_device` will work just fine, even if I pass other device flags later on.