OpenKinect/libfreenect #582: No module found named FREENECT

jithendrabsy, 2019-05-29:
jithendra@kaneki:~/libfreenect-master/wrappers/python$ python kinect_test.py 

Traceback (most recent call last):
  File "kinect_test.py", line 2, in <module>
    import freenect
ModuleNotFoundError: No module named 'freenect'

libasoles, 2019-05-30:
Check you python version. maybe you installed freenect for python2 instead of python3. That happens when you have pip and pip3, and pip refers to python2.

Quick check: 
pip --version

If you think this has to do with that, please post your commands (the ones you used to install the modules) so we can check them.

jithendrabsy, 2019-05-31:
I think it's the problem with my 1473 model. 
" Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO "
I'm always getting this error?
Can I able to use this model for Gaze detection?

piedar, 2019-06-04:
Another user [reported](https://github.com/OpenKinect/libfreenect/issues/543#issuecomment-492024478) that removing [this line of code](https://github.com/OpenKinect/libfreenect/blob/aca6046a5bc1bc5bbe7f2ae8c0085464e7ebd9f1/src/usb_libusb10.c#L443) fixes the problem with `LIBUSB_ERROR_IO`.  Maybe that will help you too.