OpenKinect/libfreenect #587: No callbacks for 0.5.6 on osx model 1517

SoylentGraham, 2019-06-12:
Installing from homebrew, 0.5.1 works, but not 0.5.6.
Using a 1517 model.

[with 0.5.6] The device is detected, opened, enabled depth streaming, but never get a callback. `freenect_process_events()` reports no errors.

Switched back to 0.5.1 (as I had it to hand) and with no code changes, I'm getting depth callbacks.

Incidentally the demos don't run successfully either either missing glut (which is probably a homebrew problem), or just no error (cam test), but that may be misleading.

Not investigated further, but reporting so I don't forget about it.

OSX is `10.13.6` (high sierra) on a 2011 imac (iMac12,2)

SoylentGraham, 2019-06-13:
This device also didn't give me any frames back on windows with the current master  (building source directly in my project) [7cb0d3c3c43c9cc25a39492668858fd554c46e99]

Tag 5.6 also doesn't give me any frames. (Both find the device, although serial is 000000000 as opposed to the proper one on OSX, but I guess that may be unrelated, or something to do with the libusbk driver)

Reset to 5.5 tag and it works.

piedar, 2019-06-14:
Hmm, looking at the [diff between v0.5.5 and v0.5.6](https://github.com/OpenKinect/libfreenect/compare/v0.5.5...v0.5.6?diff=split) the biggest change is in usb_libusb10.c and mostly related to opening devices.  I've heard that the led keep alive code can cause issues, so you could try commenting out the [call to `fnusb_keep_alive_led()`](https://github.com/OpenKinect/libfreenect/blob/db0e1825ff93133f3ca5735524d78279ede824db/src/usb_libusb10.c#L443).