OpenKinect/libfreenect #590: Can you let libfreenect support kinect for Azure?

nonlinear1, 2019-07-12:
Recently, I got a kinect for azure from microsoft company. I want let this new kinect run on nvidia Xavier, I failed. The reason is that microsoft no arm archeture sdk for ubuntu. I try compile sdk sources downloaded from github for arm archeture. I also failed, because the depthEngine is close source.
So, could you someone give me suggestion and I want code for a new depthEngine for arm archeture on ubuntu 18.04.

floe, 2019-07-16:
There's two big chunks that need to be done for an open source driver for Kinect Azure. First, we need to capture and analyze the USB traffic from the device to figure out the communication protocol, and second, we need to reverse engineering the phase unwrapping algorithm. If you can provide a wireshark capture of USB traffic, that would be a great help.

nonlinear1, 2019-07-22:
Thanks for your reply. Could you give me some specific instructions and I will help you add support for kinect for Azure

floe, 2019-07-22:
First, you'll need to install Wireshark and setup USB capture (specific instructions see Wireshark wiki).
Next step is to start the capture, then connect the Kinect Azure, stream some depth & color data using the standard Microsoft tools, close the session and the save the capture data as a PCAP file.

nonlinear1, 2019-07-23:
OK，I will try, then I will send the PCAP file to your e-mail:floe@butterbrot.org

floe, 2019-07-24:
Excellent, thank you very much. With your permission, I will upload the file to the libfreenect2 repo so that others can have a look, too.

floe, 2019-07-24:
Let's move discussion to https://github.com/OpenKinect/libfreenect2/issues/1067.

closed 2019-07-24 by floe

nonlinear1, 2019-07-24:
OK，you can upload the file