OpenKinect/libfreenect #591: make failing on MacOS 10.15 beta 3 (Catalina)

virusys, 2019-07-15:
I recently upgraded to a new machine (2018 Mac Mini) running the 10.15 beta for another project I'm working on. Running make generates this result:

```
$ make
[  1%] Building C object src/CMakeFiles/freenectstatic.dir/core.c.o
/Users/billjobs/code/libfreenect/src/core.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/freenectstatic.dir/core.c.o] Error 1
make[1]: *** [src/CMakeFiles/freenectstatic.dir/all] Error 2
make: *** [all] Error 2
```
Apparently Apple has restructured the include directories to be all self contained within the Xcode.app package, I have tried to point make to the Xcode toolchain's header files with this command but got the same result:

```
$ make -I/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1
[  1%] Building C object src/CMakeFiles/freenectstatic.dir/core.c.o
/Users/billjobs/code/libfreenect/src/core.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/freenectstatic.dir/core.c.o] Error 1
make[1]: *** [src/CMakeFiles/freenectstatic.dir/all] Error 2
make: *** [all] Error 2
```

I'm not sure if this is necessarily a bug or just me not fully understanding the intricacies of building command line software, but if someone could help point me in the right direction I would greatly appreciate it!

piedar, 2019-07-15:
I think the custom path should apply during cmake instead.  You could try

```
cmake .. -DCMAKE_C_FLAGS="-I/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1"
make
```

virusys, 2019-07-15:
Thanks for that suggestion — after entering those commands I get this result:

```
$ make
[  1%] Building C object src/CMakeFiles/freenectstatic.dir/core.c.o
In file included from /Users/billjobs/code/libfreenect/src/core.c:27:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdio.h:108:15: fatal error: 'stdio.h' file not found
#include_next <stdio.h>
              ^~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/freenectstatic.dir/core.c.o] Error 1
make[1]: *** [src/CMakeFiles/freenectstatic.dir/all] Error 2
make: *** [all] Error 2
```

I guess the headers in Xcode's toolchain are the LLVM specific ones, and the `#include_next <stdio.h>` is still looking for the standard file (and can't find it). Weird!

matheusft, 2023-01-27:
I'm getting this

```
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wrappers/python/python3/freenect.so] Error 1
make[1]: *** [wrappers/python/CMakeFiles/cython3_freenect.dir/all] Error 2
make: *** [all] Error 2
(venv_identikey_38) matheustorquato@Matheuss-M1-MBP libfreenect % 
```