OpenKinect/libfreenect #595: Unable to change resolution to low using the Python3 wrapper

josuerocha, 2019-10-05:
I am trying to grab depth and RGB frames from the sensor using the 320*240 resolution. Therefore I did the following:

`context = freenect.init()
    device = freenect.open_device(context, 0)
    freenect.set_depth_mode(device, freenect.RESOLUTION_LOW, freenect.DEPTH_11BIT)
    freenect.set_video_mode(device, freenect.RESOLUTION_LOW, freenect.VIDEO_RGB)
`

However, I get the following error: 

`freenect_set_depth_mode: freenect_frame_mode provided is invalid
`

piedar, 2019-10-13:
Looks like low resolution was never implemented.  See [cameras.c](https://github.com/OpenKinect/libfreenect/blob/83e57e1318cc64c9aabac481b9e330acc1914a23/src/cameras.c#L42) for the list of video and depth modes.

josuerocha, 2019-10-15:
That explains it. Thank you for the clarification.

Are there any plans to implement low resolution?