OpenKinect/libfreenect #599: Catalina build fails with Python3 wrapper

mjsML, 2019-10-26:
"__Py_NoneStruct", referenced from:
      ___pyx_pw_8freenect_43open_device in freenect3.c.o
      ___pyx_f_8freenect_init in freenect3.c.o
      ___pyx_f_8freenect_open_device in freenect3.c.o
      ___pyx_tp_new_8freenect_DevPtr in freenect3.c.o
      ___pyx_pymod_create in freenect3.c.o
      ___pyx_pymod_exec_freenect in freenect3.c.o
      ___pyx_tp_clear_8freenect_DevPtr in freenect3.c.o
      ...
  "__Py_TrueStruct", referenced from:
      ___Pyx_AddTraceback in freenect3.c.o
      ___pyx_f_8freenect_depth_cb in freenect3.c.o
      ___pyx_f_8freenect_video_cb in freenect3.c.o
      ___pyx_pw_8freenect_49runloop in freenect3.c.o
      ___pyx_pw_8freenect_51base_runloop in freenect3.c.o
      ___pyx_pw_8freenect_53sync_get_depth in freenect3.c.o
      ___pyx_pw_8freenect_55sync_get_video in freenect3.c.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wrappers/python/python3/freenect.so] Error 1
make[1]: *** [wrappers/python/CMakeFiles/cython3_freenect.dir/all] Error 2
make: *** [all] Error 2

Obrepal, 2021-11-02:
Hi, did you manage to solve it?

matheusft, 2023-01-27:
I'm having the same problem

Obrepal, 2023-01-27:
> 

In the end I managed to solve it, but sometimes it doesnt see a sensor or pipeline is blocked.

matheusft, 2023-01-27:
How did you solve it? @Obrepal