OpenKinect/libfreenect #609: only kinect xbox nui motor is detected!!!!

predator4hack, 2020-02-16:
i am using kinect v1 on ubuntu on a virtualbox but only xbox nui motor is detected

$lsusb
Bus 001 Device 004: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

camera and audio are not detected

aryahadii, 2020-11-12:
Maybe you should disable `autosuspend` as mentioned [here](https://openkinect.org/wiki/Getting_Started#Problems_with_accessing_device).

amm98d, 2020-12-14:
> Maybe you should disable `autosuspend` as mentioned [here](https://openkinect.org/wiki/Getting_Started#Problems_with_accessing_device).

I have the same issue. I have tried disabling `autosuspend` as mentioned in the link. But still the same issue. Has anyone solved this?

TzionCastillo1, 2022-08-18:
I had the same issue. I encountered a blog post from a while ago that said this is usually caused by a power issue with the Kinect. It turns out when I modified the cable to use my own 12v supply, I reversed the polarity. Switched the power wires around and now everything works.
Hope this helps

amm98d, 2022-08-30:
> I had the same issue. I encountered a blog post from a while ago that said this is usually caused by a power issue with the Kinect. It turns out when I modified the cable to use my own 12v supply, I reversed the polarity. Switched the power wires around and now everything works. Hope this helps

Yes, it ultimately turned out to be a power-related issue. Some problem with the power supply.