OpenKinect/libfreenect #617: Can't call both sync_get_video() and open_device() in Python wrapper

temccarthy, 2020-05-20:
I wrote a program that gets the video and depth information from my kinect using sync_get_video() and sync_get_depth(), and that works perfectly fine. However, when I add
`ctx = freenect.init()
dev = freenect.open_device(ctx, 0)`
to the beginning of the program, I get `Failed to claim camera interface: LIBUSB_ERROR_BUSY`. Thus I can't access anything else for the kinect like the led or tilt motor.

piedar, 2020-05-23:
Unfortunately the sync wrapper takes full control over devices, so there's no way to use it with custom device management in the same program.

logan-tan, 2022-01-21:
@piedar Can you elaborate on this? Does this mean that I cannot modify the device settings and capture images in the same program or am I musunderstanding?

piedar, 2022-01-21:
@logan-tan The sync API and regular API both provide methods to modify settings and capture images.  The problem comes when mixing the two.  The [sync code](https://github.com/OpenKinect/libfreenect/blob/master/wrappers/c_sync/libfreenect_sync.c#L217) handles all of the setup and configuration and is meant to be used as a wrapper.  So you must choose between __only__ sync methods or __only__ non-sync methods.  (Ideally I think there should have been two separate python modules to avoid this confusion.)

The original question was about the tilt capabilities, which [can be set via the sync API](https://github.com/OpenKinect/libfreenect/blob/master/wrappers/c_sync/libfreenect_sync.c#L424) but unfortunately the python wrapper does not expose these tilt methods.