OpenKinect/libfreenect #618: use kinect mic array as system mic

shotbyapony, 2020-05-30:
title says it all - im trying to use it for alexa voice recognition but i can't seem to figure it out. my kinect is the 360 one model 1414. any help is appreciated, thanks!

piedar, 2020-06-03:
If you use Linux, try [kinect-audio-setup](https://git.ao2.it/kinect-audio-setup.git/) (packaged in [Debian contrib](https://packages.debian.org/sid/sound/kinect-audio-setup), [Arch user repo](https://aur.archlinux.org/packages/kinect-audio-setup/), [my Gentoo overlay](https://github.com/piedar/overboard/tree/master/dev-libs/kinect-audio-setup)).  It uses udev to load firmware and makes the mic available to the system as an audio device.

haxpor, 2020-06-17:
I have to install `kinect-audio-setup` as per piedar suggests. I'm on ubuntu 20.04 but I need to reboot the system to let system detects it. Thanks piedar!

closed 2021-02-11 by piedar