OpenKinect/libfreenect #624: Can't change the angle of kinect

barfin, 2020-10-03:
I connected my Kinect v1 to my arch liunx and i installed freenect everything works fine including tilt demo using "freenect-tiltdemo" command
but i don't see a way to change the angle my kinect manually
What's the command for that? i searched the wiki and didn't find anything

piedar, 2020-10-05:
In the API this is [`freenect_set_tilt_degs()`](https://github.com/OpenKinect/libfreenect/blob/ad5b98de90ccc2aaac696eaf9fb1a5965877c055/include/libfreenect.h#L526).  You can try it out with `freenect-glview` - hit w to tilt up or x to tilt down.

barfin, 2020-10-05:
> In the API this is [`freenect_set_tilt_degs()`](https://github.com/OpenKinect/libfreenect/blob/ad5b98de90ccc2aaac696eaf9fb1a5965877c055/include/libfreenect.h#L526). You can try it out with `freenect-glview` - hit w to tilt up or x to tilt down.

thanks that was exactly what i was looking for

closed 2020-10-05 by barfin