OpenKinect/libfreenect #628: fakenect in 0.6.0 is unable to replay back previously recorded footage

smokhov, 2020-10-29:
The most recent version cannot replay depth-only and color-depth-only
recordings from the versions prior. It seems has to do with absolutely
requiring a presence of an IR/registered image in the recording. The flexibility
of having any of the streams in the recording and playback of the past
recorded datasets is needed. I believe the problem was introduced
aroudn 83e57e1318cc64c9aabac481b9e330acc1914a23.

How can we avoid the registration as an option?

- Example data set (depth+color): https://bitbucket.org/openiss/public/downloads/test-recording.tar.bz2
  - It works with  f63bb6a
  - but not with 0.6.0
- another dataset there with depth only also does not work

piedar, 2021-02-11:
How can I replicate the problem?  I have tried your test-recording with the examples and it appears to work as expected.

```
LD_PRELOAD="/usr/lib64/fakenect/libfakenect.so" FAKENECT_PATH="./test-recording" freenect-glview
```

smokhov, 2021-02-12:
I need to reproduce the problem again in the environment we had, and will let you know. I was working with someone who had Ubuntu 18 I believe and the pre-packaged libfreenect 0.6.x that came from apt get was bailing out with an error missing format or some such (again, I need to reproduce the exact error we were getting, but it was clear it was looking for a registered frame that it could not find in the recording). After switching to a fork submodule I have in OpenISS and uninstalling libfreenect from Ubuntu, and recompiling it worked again with our datasets.

piedar, 2022-09-20:
Seems like you found a workaround - please reopen if there's still an issue.

closed 2022-09-20 by piedar