OpenKinect/libfreenect #630: fails to compile with python3

RadTekWeb, 2020-11-17:
As title says, fails when compiling `freenect3.c`:

```
[ 91%] Linking CXX executable ../../bin/freenect-cpp_pcview
[ 91%] Built target freenect-cpp_pcview
[ 92%] Building CXX object wrappers/cpp/CMakeFiles/freenect-cppview.dir/cppview.cpp.o
[ 94%] Linking CXX executable ../../bin/freenect-cppview
[ 94%] Built target freenect-cppview
[ 96%] Generating freenect3.c
Scanning dependencies of target cython3_freenect
[ 98%] Building C object wrappers/python/CMakeFiles/cython3_freenect.dir/freenect3.c.o
cc1: warning: /usr/bin/python3: not a directory
/home/username/libfreenect/build/wrappers/python/freenect3.c:4:10: fatal error: Python.h: No such file or directory
    4 | #include "Python.h"
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [wrappers/python/CMakeFiles/cython3_freenect.dir/build.make:67: wrappers/python/CMakeFiles/cython3_freenect.dir/freenect3.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:743: wrappers/python/CMakeFiles/cython3_freenect.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I'm on mint 20, python3.8.5, cython3 0.29.14, cmake 3.16 if that matters, 
thanks

piedar, 2021-02-11:
Have you installed the `python3-dev` package?

RadTekWeb, 2021-02-11:
> Have you installed the `python3-dev` package?

Yes, `3.8.2-0ubuntu2`

piedar, 2021-02-11:
Is there any chance you have multiple python versions installed?  For example I have 3.8 and 3.9 so I have to write

```
cmake .. -L -DBUILD_PYTHON3=ON -DPython3_EXACTVERSION=3.8
```

to make cmake use the correct version.  Then it should say something like

```
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.7") found components: Interpreter Development Development.Module Development.Embed
```

and cmake should set `Python3_INCLUDE_DIRS` properly.

RadTekWeb, 2021-02-11:
That was it, thank you so much!

closed 2021-02-11 by RadTekWeb