OpenKinect/libfreenect #639: Dropouts in data from Kinect on Raspberry Pi unless bluetooth dongle also plugged in

JesseCake, 2021-07-09:
When using libfreenect with a Kinect v1 on ROS on a Raspberry Pi 4, running either Raspbian or Ubuntu 20.04, there are huge amounts of dropouts and missed packets (bad magic etc) until I plug a small USB bluetooth dongle into another USB.

Suddenly the data throughput is perfect, doesn't miss a single packet and continues to run properly.

If I go to restart pulling data from the Kinect, I have to also re-plug the bluetooth dongle otherwise I'm back where I started.

What could be causing this? 

I learned this trick from another who also mentioned the same process to get stable readings. Here is the blog where it's discussed:
[here](https://www.hackster.io/dmitrywat/rgb-d-slam-with-kinect-on-raspberry-pi-4-ros-melodic-ace795)
(yes at the top it speaks mostly about melodic, but also goes into Noetic further down)

The blogger discusses that it needs to "load up" the usb - but what could that mean?

Obviously this isn't a great solution as I need to keep replugging this dongle each time I want to take readings from the Kinect which isn't great in a robot.

Where can I start troubleshooting? What could be the root cause of this?

**Here is my lsusb -t (with kinect and bluetooth dongle attached):**
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 13, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 1: Dev 13, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 1: Dev 13, If 2, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 1: Dev 13, If 3, Class=Application Specific Interface, Driver=, 12M
        |__ Port 2: Dev 8, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 3: Dev 12, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
            |__ Port 1: Dev 10, If 0, Class=Vendor Specific Class, Driver=, 480M
            |__ Port 2: Dev 9, If 0, Class=Vendor Specific Class, Driver=, 12M

Any hints? Of course this could be specific to the platform, but it seems odd that the fix would be to plug another dongle in?

I'd imagine that many would be keen to easily get freenect readings into ROS on Raspberry Pi because of the popular and cheap nature of the board.

Many thanks in advance

piran-prog, 2022-09-11:
I'm getting the same thing. Lots of:
[image]

plugging in an ancient bluetooth dongle helped a little with the freenect example files but introduced a lot of lag when trying to use it with something else (Orb-SLAM3 in this case).
Would wellcome any ideas on where to look for this issue?

piran-prog, 2022-09-11:
looking at the lppf numbers, it's showing arounf 30lppf without the bluetooth dongle and 14 lppf with the bluetooth dongle