OpenKinect/libfreenect #643: Python 3.8+: PyTypeObject {alias »struct _typeobject} has no member named tp_print

SoulEater45, 2021-09-17:
I tried to compile this little puppy for an old kinect lying in the office.
I am running python 3.9 and succeeded in compiling the library itself, but when trying to install the python lib I get the error mentioned in the title (something similar can be found [here: https://bugs.gentoo.org/788703](here: https://bugs.gentoo.org/788703)).
So if someone tries to compile it, use an older version. I haven't tested it, but I think it should work?

piedar, 2021-09-17:
Strangely the cmake build process does not update the freenect.c file.  You can try doing so by installing cython then running

```sh
cd wrappers/python
python setup.py build_ext --inplace
```

pashultz, 2022-03-18:
Hi! I'm seeing the same issue using python 3.9.7 and cython 0.29.24 to compile commit 5b2a20e. Using `build_ext --inplace` gives the same error. Any suggestions?

ming82871, 2023-03-13:
in .cpp file  you can change tp_print -> tp_dict

piedar, 2023-04-03:
The existing `freenect.c` is outdated so it's better to generate it locally using one of these methods.

- run `python setup.py build_ext -f`
- delete `wrappers/python/freenect.c` (as done in 8f5ba392ddcfb62df31c3d5342c0989cbad7233f) and try again
- use `cmake -DBUILD_PYTHON3=ON` rather than running `setup.py`