OpenKinect/libfreenect #647: Libfreenect Solution Build Error-Windows

senayapsu, 2021-12-04:
From tihs github page and this page(https://openkinect.org/wiki/Getting_Started), I followed the download steps for libfreenect but in the solution build in visual studio I have this error. There is no libusb.h file found in the project. However, It should exist.

[image]

Then on the web, I found a libusb.h file myself and put it inside the project. After that it see libusb.h file, but errors increased.

How can I solve these errors and build the solution for libfreenect?

[image]

dcnavarros-intecol, 2022-05-20:
I'm Getting exactly the same kind of problems, did you solved it?

GameDungeon, 2022-10-15:
I am also getting the same. As far as I can tell, libusb doesn't come packaged with a header file (only a DLL and lib) on windows. It only does for linux. This poses an issue as libfreenect asks for a header file/