OpenKinect/libfreenect #648: Build errors on linux

jwillb, 2021-12-13:
Hi.
I have recently wanted to start developing applications for my Kinect, so I cloned this library and tried to follow the buiid instructions.
```cmake -L .. runs``` without a hitch, and creates a Makefile. It says I have all of the required libraries (including glut) and lists the options.
Then, I run make. This does not work as intended and it appears as though the library is not properly linked.
```
[  1%] Building C object src/CMakeFiles/freenect.dir/core.c.o
[  3%] Building C object src/CMakeFiles/freenect.dir/tilt.c.o
[  5%] Building C object src/CMakeFiles/freenect.dir/cameras.c.o
/home/jonathanb/Repositories/libfreenect/src/cameras.c:74:34: warning: ‘invalid_mode’ defined but not used [-Wunused-const-variable=]
   74 | static const freenect_frame_mode invalid_mode = {0, (freenect_resolution)0, {(freenect_video_format)0}, 0, 0, 0, 0, 0, 0, 0};
      |                                  ^~~~~~~~~~~~
[  7%] Building C object src/CMakeFiles/freenect.dir/flags.c.o
[  9%] Building C object src/CMakeFiles/freenect.dir/usb_libusb10.c.o
[ 11%] Building C object src/CMakeFiles/freenect.dir/registration.c.o
[ 13%] Building C object src/CMakeFiles/freenect.dir/audio.c.o
[ 15%] Building C object src/CMakeFiles/freenect.dir/loader.c.o
/home/jonathanb/Repositories/libfreenect/src/loader.c:82:12: warning: ‘check_version_string’ defined but not used [-Wunused-function]
   82 | static int check_version_string(fnusb_dev* dev) {
      |            ^~~~~~~~~~~~~~~~~~~~
[ 16%] Linking C shared library ../lib/libfreenect.so
[ 16%] Built target freenect
[ 18%] Building C object src/CMakeFiles/freenectstatic.dir/core.c.o
[ 20%] Building C object src/CMakeFiles/freenectstatic.dir/tilt.c.o
[ 22%] Building C object src/CMakeFiles/freenectstatic.dir/cameras.c.o
/home/jonathanb/Repositories/libfreenect/src/cameras.c:74:34: warning: ‘invalid_mode’ defined but not used [-Wunused-const-variable=]
   74 | static const freenect_frame_mode invalid_mode = {0, (freenect_resolution)0, {(freenect_video_format)0}, 0, 0, 0, 0, 0, 0, 0};
      |                                  ^~~~~~~~~~~~
[ 24%] Building C object src/CMakeFiles/freenectstatic.dir/flags.c.o
[ 26%] Building C object src/CMakeFiles/freenectstatic.dir/usb_libusb10.c.o
[ 28%] Building C object src/CMakeFiles/freenectstatic.dir/registration.c.o
[ 30%] Building C object src/CMakeFiles/freenectstatic.dir/audio.c.o
[ 32%] Building C object src/CMakeFiles/freenectstatic.dir/loader.c.o
/home/jonathanb/Repositories/libfreenect/src/loader.c:82:12: warning: ‘check_version_string’ defined but not used [-Wunused-function]
   82 | static int check_version_string(fnusb_dev* dev) {
      |            ^~~~~~~~~~~~~~~~~~~~
[ 33%] Linking C static library ../lib/libfreenect.a
[ 33%] Built target freenectstatic
[ 35%] Building C object examples/CMakeFiles/freenect-camtest.dir/camtest.c.o
[ 37%] Linking C executable ../bin/freenect-camtest
[ 37%] Built target freenect-camtest
[ 39%] Building C object examples/CMakeFiles/freenect-wavrecord.dir/wavrecord.c.o
[ 41%] Linking C executable ../bin/freenect-wavrecord
[ 41%] Built target freenect-wavrecord
[ 43%] Building C object examples/CMakeFiles/freenect-glview.dir/glview.c.o
[ 45%] Linking C executable ../bin/freenect-glview
/usr/bin/ld: CMakeFiles/freenect-glview.dir/glview.c.o: in function `keyPressed':
glview.c:(.text+0x88f): undefined reference to `glutDestroyWindow'
/usr/bin/ld: CMakeFiles/freenect-glview.dir/glview.c.o: in function `gl_threadfunc':
glview.c:(.text+0x1053): undefined reference to `glutInit'
/usr/bin/ld: glview.c:(.text+0x105d): undefined reference to `glutInitDisplayMode'
/usr/bin/ld: glview.c:(.text+0x106c): undefined reference to `glutInitWindowSize'
/usr/bin/ld: glview.c:(.text+0x1075): undefined reference to `glutInitWindowPosition'
/usr/bin/ld: glview.c:(.text+0x1081): undefined reference to `glutCreateWindow'
/usr/bin/ld: glview.c:(.text+0x108f): undefined reference to `glutDisplayFunc'
/usr/bin/ld: glview.c:(.text+0x1097): undefined reference to `glutIdleFunc'
/usr/bin/ld: glview.c:(.text+0x10a3): undefined reference to `glutReshapeFunc'
/usr/bin/ld: glview.c:(.text+0x10af): undefined reference to `glutKeyboardFunc'
/usr/bin/ld: glview.c:(.text+0x10c3): undefined reference to `glutMainLoop'
/usr/bin/ld: CMakeFiles/freenect-glview.dir/glview.c.o: in function `DrawGLScene':
glview.c:(.text+0x38b): undefined reference to `glutSwapBuffers'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/freenect-glview.dir/build.make:101: bin/freenect-glview] Error 1
make[1]: *** [CMakeFiles/Makefile2:327: examples/CMakeFiles/freenect-glview.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
Since this was just an error with the glview example program, I tried running ```cmake .. -BUILD_EXAMPLES:OFF``` and got this error message instead (I removed the CMakeCache File):
```
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Operating system is Linux
-- Got System Processor x86_64
-- Linux x86_64 Detected
-- libfreenect will be installed to /usr/local
-- Headers will be installed to /usr/local/include/libfreenect
-- Libraries will be installed to /usr/local/lib
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/libusb-1.0.so
-- Found PythonInterp: /usr/bin/python (found version "3.9.9") 
-- Looking for include file pthread.h
-- Looking for include file pthread.h - not found
-- Could NOT find Threads (missing: Threads_FOUND) 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  examples/CMakeLists.txt:16 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/libOpenGL.so   
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.0") 
-- Checking for module 'glut'
--   Found glut, version 3.2.1
-- Found GLUT: TRUE  
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindThreads.cmake:245 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  wrappers/c_sync/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jonathanb/Repositories/libfreenect/build/UILD_EXAMPLES:OFF/CMakeFiles/CMakeOutput.log".
See also "/home/jonathanb/Repositories/libfreenect/build/UILD_EXAMPLES:OFF/CMakeFiles/CMakeError.log".
```
It appears that threads is now somehow missing. Any ideas?
(I am running a GeForce GTX 10603GB, with an Intel i7-8700 on Artix Linux. I suspect this could be a problem with the distro because I have successfully used this library on another distro on the same hardware.)
Any and all help is appreciated.
Thanks!

jwillb, 2021-12-15:
Turns out, uninstalling pkgconfig (```pkgconf``` is the package name on Artix Linux at least) fixes the problem. However, I'll leave this issue open as this does not seem like a scalable solution for everyone.

prayujt, 2022-09-06:
Anyone fixed this issue yet? Uninstalling pkgconfig didn't make any difference for me, and I'm having the same error listed above.

piedar, 2022-09-19:
Likely fixed by #663

closed 2022-09-19 by piedar