OpenKinect/libfreenect #649: Allow to perform registration while keeping the depth buffer

SirLynix, 2021-12-29:
Hi, author of [obs-kinect](https://github.com/SirLynix/obs-kinect) here.

First of all, thank you for your amazing job on libfreenect, I've been using it for my plugin but there's one issue that remains, I need to be able to get registered depth (color-mapped depth as I call it) and regular depth at the same time, however libfreenect seems to be the only kinect backend which doesn't provide a function to map depth to rgb space (libfreenect2 has `Registration::apply`, Kinect SDKs have `MapColorFrameToDepthFrame`, Azure SDK has [k4a_transformation_depth_image_to_color_camera](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/refs/pull/1085/merge/group___functions_gafacffb5f781a9c2df30d4a16241cd514.html#gafacffb5f781a9c2df30d4a16241cd514)).

I see there's the internal `freenect_apply_depth_to_mm` which seems to do what I want, is there a way it could be exposed?

Thank you.

piedar, 2022-01-03:
I think you might want `freenect_apply_registration` because it aligns the result to the RGB image, while `freenect_apply_depth_to_mm` converts to mm but does not align.

SirLynix, 2022-01-04:
Yes indeed. I made a fork of libfreeenect to [expose such features](https://github.com/OpenKinect/libfreenect/compare/master...SirLynix:master) which [allowed me to have both depth and color-mapped depth images](https://github.com/SirLynix/obs-kinect/commit/5e67696e7ebe8ec0839899268d12d1e3cf2390b1).

piedar, 2022-01-04:
Nice!  I had a [quick go](https://github.com/OpenKinect/libfreenect/compare/public-registration) at it the other day.  It's a bit more complicated, but I kept `convert_packed_to_16bit` internal and my registration function handles all the depth modes rather than just `FREENECT_DEPTH_11BIT`.  Let me know what you think.

SirLynix, 2022-01-04:
Looks good! I will give it a try to check if it works as expected soon, thank you!

edugimeno, 2022-04-18:
Please please forgive me for hijacking this thread for this question, but there's absolutely no place / forum I can find to place a couple questions about this project. The wiki links to google groups which were closed 4 years ago. There's no other trace of forum for OpenKinect, can you please help me with directions for any currently working forum for OpenKinect / Libfreenect?

Thanks in advance and sorry

piedar, 2022-09-19:
@edugimeno Alas, there isn't really a forum that I know of.  I check in here on GitHub occasionally to fix bugs and give suggestions, but most contributors have moved on to other projects.

commit c8c6716f (2022-01-04): feat: expose a function to apply depth registration - fixes #649
commit 5c54a0b2 (2022-01-04): feat: expose a function to apply depth registration - fixes #649