OpenKinect/libfreenect #651: ERROR TO MAKE

lvnlinh1812, 2022-02-05:
I run "make" in directory "build" but having this error. Anyone can help me?
[image]

michaelshiel, 2022-02-20:
I had this error, I think it's because newer versions of GLUT/cmake changed something. I don't know CMake well enough to say, but if you change `${GLUT_LIBRARY}` to `GLUT::GLUT` (patch below) it built fine for me.

```
diff --git a/examples/CMakeLists.txt b/examples/CMakeLists.txt
index 8c6733f..d79f56e 100644
--- a/examples/CMakeLists.txt
+++ b/examples/CMakeLists.txt
@@ -25,11 +25,11 @@ if (Threads_FOUND AND OPENGL_FOUND AND GLUT_FOUND)
   add_executable(freenect-chunkview chunkview.c)
   add_executable(freenect-micview micview.c)
 
-  target_link_libraries(freenect-glview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
-  target_link_libraries(freenect-regview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
-  target_link_libraries(freenect-hiview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
-  target_link_libraries(freenect-chunkview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
-  target_link_libraries(freenect-micview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+  target_link_libraries(freenect-glview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+  target_link_libraries(freenect-regview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+  target_link_libraries(freenect-hiview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+  target_link_libraries(freenect-chunkview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+  target_link_libraries(freenect-micview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
 
   install(TARGETS freenect-glview freenect-regview freenect-hiview freenect-chunkview freenect-micview
           DESTINATION bin)
@@ -51,7 +51,7 @@ if (BUILD_C_SYNC)
   if (OPENGL_FOUND AND GLUT_FOUND)
     include_directories(${OPENGL_INCLUDE_DIR} ${GLUT_INCLUDE_DIR})
     add_executable(freenect-glpclview glpclview.c)
-    target_link_libraries(freenect-glpclview freenect_sync ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${MATH_LIB})
+    target_link_libraries(freenect-glpclview freenect_sync ${OPENGL_LIBRARIES} GLUT::GLUT ${MATH_LIB})
     install(TARGETS freenect-glpclview
             DESTINATION bin)
   endif ()
diff --git a/wrappers/cpp/CMakeLists.txt b/wrappers/cpp/CMakeLists.txt
index 6d17924..ee8670b 100644
--- a/wrappers/cpp/CMakeLists.txt
+++ b/wrappers/cpp/CMakeLists.txt
@@ -18,8 +18,8 @@ if (BUILD_EXAMPLES)
     target_compile_features(freenect-cppview PUBLIC cxx_std_11)
     target_compile_features(freenect-cpp_pcview PUBLIC cxx_std_11)
 
-    target_link_libraries(freenect-cppview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
-    target_link_libraries(freenect-cpp_pcview freenect ${OPENGL_LIBRARIES} ${GLUT_LIBRARY} ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+    target_link_libraries(freenect-cppview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
+    target_link_libraries(freenect-cpp_pcview freenect ${OPENGL_LIBRARIES} GLUT::GLUT ${CMAKE_THREAD_LIBS_INIT} ${MATH_LIB})
 
     install(TARGETS freenect-cppview freenect-cpp_pcview
             DESTINATION bin)
```

piedar, 2022-09-19:
Likely fixed by #663

closed 2022-09-19 by piedar