OpenKinect/libfreenect #655: Can't build libfreenect on Windows 10/11 "LNK1120: 8 unresolved externals"

DunZek, 2022-04-06:
I am unable to build libfreenect on Windows 10 and 11, which I believe, is due to some problems with pthreads. 

Within `C:\ComputerStuff\libfreenect\build`, I ran and configured cmake as follows:

```ps
> cmake -L ..
  -DTHREADS_PTHREADS_WIN32_LIBRARY=C:/ComputerStuff/PTHREADS-BUILT/lib/pthreadVC3.lib 
  -DTHREADS_PTHREADS_INCLUDE_DIR=C:/ComputerStuff/PTHREADS-BUILT/include 
  -DLIBUSB_1_INCLUDE_DIR=C:/ComputerStuff/libusb-1.0.25/include/libusb-1.0 
  -DLIBUSB_1_LIBRARY=C:/ComputerStuff/libusb-1.0.25/VS2019/MS64/Debug/lib/libusb-1.0.lib
  -DBUILD_PYTHON3=ON 
```

Which produced this output:
```
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22572.
-- Operating system is Windows
-- Got System Processor AMD64
-- libfreenect will be installed to C:/Program Files (x86)/libfreenect
-- Headers will be installed to C:/Program Files (x86)/libfreenect/include/libfreenect
-- Libraries will be installed to C:/Program Files (x86)/libfreenect/lib
-- Found libusb-1.0:
--  - Includes: C:/ComputerStuff/libusb-1.0.25/include/libusb-1.0
--  - Libraries: C:/ComputerStuff/libusb-1.0.25/VS2019/MS64/Debug/dll/libusb-1.0.lib
-- Checking for module 'glut'
--   No package 'glut' found
-- Could NOT find GLUT (missing: GLUT_glut_LIBRARY GLUT_INCLUDE_DIR)
-- Checking for module 'glut'
--   No package 'glut' found
-- Could NOT find GLUT (missing: GLUT_glut_LIBRARY GLUT_INCLUDE_DIR)
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ComputerStuff/libfreenect/build
-- Cache values
BUILD_AS3_SERVER:BOOL=OFF
BUILD_CPP:BOOL=ON
BUILD_CV:BOOL=OFF
BUILD_C_SYNC:BOOL=ON
BUILD_EXAMPLES:BOOL=ON
BUILD_FAKENECT:BOOL=ON
BUILD_OPENNI2_DRIVER:BOOL=OFF
BUILD_PYTHON:BOOL=OFF
BUILD_PYTHON2:BOOL=OFF
BUILD_PYTHON3:BOOL=ON
BUILD_REDIST_PACKAGE:BOOL=ON
CMAKE_CONFIGURATION_TYPES:STRING=Debug;Release;MinSizeRel;RelWithDebInfo
CMAKE_INSTALL_PREFIX:PATH=C:/Program Files (x86)/libfreenect
CYTHON_EXECUTABLE:FILEPATH=CYTHON_EXECUTABLE-NOTFOUND
LIBUSB_1_INCLUDE_DIR:PATH=C:/ComputerStuff/libusb-1.0.25/include/libusb-1.0
LIBUSB_1_LIBRARY:FILEPATH=C:/ComputerStuff/libusb-1.0.25/VS2019/MS64/Debug/dll/libusb-1.0.lib
Python2_EXACTVERSION:STRING=
Python3_EXACTVERSION:STRING=
```

Then, I tried to build with `cmake --build .` and produced these errors and outputs:
```
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  freenect.vcxproj -> C:\ComputerStuff\libfreenect\build\lib\Debug\freenect.dll
     Creating library C:/ComputerStuff/libfreenect/build/lib/Debug/freenect_sync.lib and object C:/ComputerStuff/libfreenect/build/lib/Debug/freenect_sync.exp
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_create referenced in function setup_kinect [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_join referenced in function freenect_sync_stop [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_mutex_init referenced in function setup_kinect [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_mutex_lock referenced in function init [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_mutex_unlock referenced in function freenect_sync_camera_to_world [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_cond_init referenced in function setup_kinect [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_cond_wait referenced in function init [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
libfreenect_sync.obj : error LNK2019: unresolved external symbol pthread_cond_signal referenced in function pending_runloop_tasks_dec [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
C:\ComputerStuff\PTHREADS-BUILT\lib\pthreadVC3.lib : warning LNK4272: library machine type 'x86' conflicts with target machine type 'x64' [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
C:\ComputerStuff\libfreenect\build\lib\Debug\freenect_sync.dll : fatal error LNK1120: 8 unresolved externals [C:\ComputerStuff\libfreenect\build\wrappers\c_sync\freenect_sync.vcxproj]
```

- Any help would be appreciated. I'm genuinely lost as to what I could be doing wrong.
- I built [pthreads](http://sourceforge.net/projects/pthreads4w/) from source `C:\ComputerStuff\pthreads4w-code-v3.0.0` using `nmake all install`
- I'm using libusb-1.0.25 and got it as the [Latest Windows Binaries](https://[libusb](https://libusb.info/).info/)

DunZek, 2022-04-08:
I used nmake to build pthreads-4-win. I'm using Visual Studio 2019 to build with cmake.

KaliKaloo, 2023-04-14:
Hi, I'm getting the same errors, did you have any luck in fixing this? Thanks.

bmobio, 2024-08-12:
Hello Dunzek, I got the same kind of errors. Did you find a way to fix it ?