OpenKinect/libfreenect #662: Distance issues with Kinect and R Pi 4

piran-prog, 2022-09-11:
Along with the other issues I'm having I've been trying the regview and I'm getting things like this:
[image]
The foreground is lighter and then shades to dark blue over about 2 meteres and then just shows the RGB image but then further away starts showing distance colours again

piedar, 2022-09-20:
I think this is a limitation of the depth shading algorithm in `freenect-regview` but I'm not sure how to fix it.  Ideally there should be a smooth color range from near to far.  As far as I can tell, the actual depth values are correct.

piran-prog, 2022-09-21:
Yes looks like an issue with regview, some of the other test apps seem to get it right. I managed (with a lot of effort) to get it working in ROS and that works but I'm trying to get it working with ORBSLAM3 and it's getting some of the depth wrong there: 
[image]
where the upper cloud of points is all on a single plane but should actually be at the front with the other points.