OpenKinect/libfreenect #673: libfreenect fails to build with Cython 3.0.0

prmehta24, 2023-07-27:
I am on Ubuntu 18.04, Cython 3.0.0, Python 3.8.

I tried to compile libfreenect with 

cmake .. -L -DBUILD_PYTHON3=ON -DPython3_EXACTVERSION=3.8 

make

It gave the error:

Cannot assign type 'void (freenect_device *, void *, uint32_t) except * nogil'

I downgraded Cython to 0.29.14 and it compiled perfectly.

closed 2024-01-01 by piedar

lulu110997, 2024-03-01:
I am using Cython 0.29.14.  [Commit 5341ff6](https://github.com/OpenKinect/libfreenect/commit/5341ff68aa946a11ac0df3b050f2524d7c51a825) didn't work for me. I have to remove the 'noexcept' otherwise i get the error below when trying to install the python wrapper

$ sudo python setup.py install
running install
running build
running build_ext
cythoning freenect.pyx to freenect.c
/usr/lib/python3/dist-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /home/rosserver/Git/libfreenect/wrappers/python/freenect.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)

Error compiling Cython file:
------------------------------------------------------------
...
    dev_out.ctx = ctx
    return dev_out

_depth_cb, _video_cb = None, None

cdef void depth_cb(freenect_device *dev, void *data, uint32_t timestamp) noexcept with gil:
                                                                        ^
------------------------------------------------------------

freenect.pyx:324:73: Syntax error in C variable declaration
building 'freenect' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c freenect.c -o build/temp.linux-x86_64-3.8/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /usr/lib/python3/dist-packages/numpy/core/include
freenect.c:1:2: error: #error Do not use this file, it is the result of a failed Cython compilation.
    1 | #error Do not use this file, it is the result of a failed Cython compilation.
      |  ^~~~~
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

piedar, 2024-03-05:
@lulu110997 Unfortunately, the `setup.py` is outdated.  The recommended way to build with the python extension is

```sh
cd libfreenect
mkdir build && cd build
cmake .. -DBUILD_PYTHON3=ON
cmake --build
```

This works because the cmake script has [a workaround for older versions of cython](https://github.com/OpenKinect/libfreenect/blob/master/wrappers/python/CMakeLists.txt#L34).  If you prefer to use `setup.py` with this cython, you must change the referenced file to `freenect.cython0.pyx`.

commit 5341ff68 (2024-01-01): fix(python): add noexcept to callbacks to fix #673