OpenKinect/libfreenect #676: Python3 wrapper, c1: fatal error C1083: Cannot open source file: 'freenect.c': No such file or directory

hugowhite, 2023-08-13:
I am trying to install the python3 wrapper I believe I have freenect installed correctly as the test exes work correctly but when I open the wrappers/python/setup.py this error comes up. I believe its because its trying to call it based on the assumption I have the python wrapper. Do I need to rebuild with the original python wrapper aswell as the python 3 one or do I need to edit setup.py to work with calling freenect3.c instead of freenect.c as that is the only thing I can think of as I do not have a freenect.c file. Any help would be greatly appreciated I have been slowly working through the install for the past few days but this step is really stumping me.

MysticalApple, 2023-12-10:
Has any solution been found yet? I'm having the same issue.

piedar, 2024-01-01:
I think the `setup.py` file was never really updated.  I build the python wrapper via cmake like this.

```
mkdir build && cd build
cmake .. -L -DBUILD_PYTHON3=ON
make
```

hbmartin, 2025-05-14:
I've been able to build both ways see: https://github.com/hbmartin/libfreenect/blob/master/.github/workflows/build.yml