OpenKinect/libfreenect #677: CMake warning  about legacy GL library and GLVND libraries for OpenGL and GLX.

Htet-Wai-Yan-HWY, 2024-01-01:
I am trying to use xbox 360 depth camera in ubuntu 22.04. When I build this project , cmake show following warning and error. 

`zyme@dev:~/libfreenect/build$ cmake -L ..
CMake Deprecation Warning at CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Operating system is Linux
-- Got System Processor x86_64
-- Linux x86_64 Detected
-- libfreenect will be installed to /usr/local
-- Headers will be installed to /usr/local/include/libfreenect
-- Libraries will be installed to /usr/local/lib
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  examples/CMakeLists.txt:55 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'glut'
--   No package 'glut' found
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  wrappers/cpp/CMakeLists.txt:25 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'glut'
--   No package 'glut' found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zyme/libfreenect/build
-- Cache values
BUILD_AS3_SERVER:BOOL=OFF
BUILD_CPACK_DEB:BOOL=OFF
BUILD_CPACK_RPM:BOOL=OFF
BUILD_CPACK_TGZ:BOOL=OFF
BUILD_CPP:BOOL=ON
BUILD_CV:BOOL=OFF
BUILD_C_SYNC:BOOL=ON
BUILD_EXAMPLES:BOOL=ON
BUILD_FAKENECT:BOOL=ON
BUILD_OPENNI2_DRIVER:BOOL=OFF
BUILD_PYTHON:BOOL=OFF
BUILD_REDIST_PACKAGE:BOOL=ON
CMAKE_BUILD_TYPE:STRING=
CMAKE_INSTALL_PREFIX:PATH=/usr/local
LIBUSB_1_INCLUDE_DIR:PATH=/usr/include/libusb-1.0
LIBUSB_1_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libusb-1.0.so
`

I have **glut** library under _"/usr/include/GL"_   but it shows  **No package 'glut' found**

piedar, 2024-01-01:
```
CMake Deprecation Warning at CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
```

This should be fixed by #678.

```
  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
```

This one doesn't make sense to me.  We've been setting `OpenGL_GL_PREFERENCE GLVND` since f6de60f5291258920ca7d03e8d593f1bab3f7867 more than a year ago.  Maybe #678 will help anyway...

```
-- Checking for module 'glut'
--   No package 'glut' found
```

This one I have no idea.  It works on my machine where the freeglut package provides these files.

```
/usr/include/GL/freeglut.h
/usr/include/GL/freeglut_ext.h
/usr/include/GL/freeglut_std.h
/usr/include/GL/freeglut_ucall.h
/usr/include/GL/glut.h
/usr/lib64/cmake/FreeGLUT/FreeGLUTConfig.cmake
/usr/lib64/cmake/FreeGLUT/FreeGLUTConfigVersion.cmake
/usr/lib64/cmake/FreeGLUT/FreeGLUTTargets-relwithdebinfo.cmake
/usr/lib64/cmake/FreeGLUT/FreeGLUTTargets.cmake
/usr/lib64/libglut.so -> libglut.so.3
/usr/lib64/libglut.so.3 -> libglut.so.3.12.0
/usr/lib64/libglut.so.3.12.0
/usr/lib64/pkgconfig/freeglut.pc
/usr/lib64/pkgconfig/glut.pc
```

parkerlreed, 2024-06-03:
freeglut3-dev does not include cmake files. Where is that coming from?

```
[parker@rogally build]$ dpkg -L freeglut3-dev
/.
/usr
/usr/include
/usr/include/GL
/usr/include/GL/freeglut.h
/usr/include/GL/freeglut_ext.h
/usr/include/GL/freeglut_std.h
/usr/include/GL/glut.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libglut.a
/usr/share
/usr/share/doc
/usr/share/doc/freeglut3-dev
/usr/share/doc/freeglut3-dev/copyright
/usr/share/doc/freeglut3-dev/download.html
/usr/share/doc/freeglut3-dev/freeglut.html
/usr/share/doc/freeglut3-dev/freeglut_logo.png
/usr/share/doc/freeglut3-dev/freeglut_user_interface.html
/usr/share/doc/freeglut3-dev/index.html
/usr/share/doc/freeglut3-dev/ogl_sm.png
/usr/share/doc/freeglut3-dev/progress.html
/usr/share/doc/freeglut3-dev/structure.html
/usr/lib/x86_64-linux-gnu/libglut.so
/usr/share/doc/freeglut3-dev/changelog.Debian.gz
```

parkerlreed, 2024-06-03:
They are provided by libglut-dev which only exists for mantic and above. This is for some reason left out from jammy and below.