OpenKinect/libfreenect #681: Questions about timestamp

goldbird5, 2024-01-10:
Hi, thanks for the great library!
I'm trying to use python wrapper as follow.
```depth, timestamp1 = freenect.sync_get_depth()
rgb, timestamp2 = freenect.sync_get_video()```
I want them to synchronized i.e. timestamp1 == timestamp2, but they appear slightly different.
Can I presume they are synchronized? If so, which timestamp should I use? Or if not, is there a way to get synchronized rgb & depth?

Besides, I wonder what's exactly the meaning of ```uint32_t``` type timestamp value. How can I convert it to ```float``` type value in 'second' scale?

Thank you.

piedar, 2024-01-13:
As [part of the data protocol](https://openkinect.org/wiki/Protocol_Documentation#Cameras), each [camera data packet](https://github.com/OpenKinect/libfreenect/blob/master/src/cameras.c#L214) from the Kinect contains a timestamp of the current frame.

A helpful contributor [documented the nature of these timestamps](https://github.com/OpenKinect/libfreenect/pull/465/files) a while back.  The clock runs at 60MHz so you should expect to see the timestamps increment by 60,000,000 every second.  It would be reasonable to assume rgb and depth frames are synchronized if they are within about 120,000 ticks (~2 milliseconds) of each other.  If you want to convert to seconds elapsed, you'll also need some similar code to remember the last value and handle the case where the timestamp loops back around.

```python
import freenect

timestamp_prev = None
ticks_elapsed = 0
ticks_per_second = 60000000
uint32_max = (1 << 32) - 1

try:
  while True:
    depth, timestamp = freenect.sync_get_depth()

    timestamp_prev = timestamp_prev or timestamp
    if timestamp < timestamp_prev:
      # timestamp overflowed uint32_t
      remainder = uint32_max - timestamp_prev
      delta = timestamp + remainder
    else:
      delta = timestamp - timestamp_prev

    timestamp_prev = timestamp
    ticks_elapsed += delta

    seconds_elapsed = ticks_elapsed / ticks_per_second
    print(timestamp, "\t", ticks_elapsed, "\t", seconds_elapsed)
finally:
  freenect.sync_stop()

```

goldbird5, 2024-01-13:
I got it. Thank you :)

closed 2024-01-13 by goldbird5