OpenKinect/libfreenect #683: Can't import python3 wrapper (ModuleNotFoundError: No module named 'freenect')

Looki2000, 2024-03-05:
I'm trying to build libfreenect wrapper for python 3 on Raspberry OS (pi 3 B) but I can't import it. (freenect-glview is working fine) Running python3 with sudo and restarting the terminal does not help. freenect is not present in pip list
I have tried to follow the official build Instructions and [this](https://github.com/sjtosco/freenect-python3-howto) (I had to set DCYTHON_EXECUTABLE to /usr/local/bin/cython).
How can I fix that?

Looki2000, 2024-03-06:
I managed to make it work by copying ``freenect.so`` into every possible site-packages and dist-packages locations of python3/3.9 and it started to work. Sadly, it's not very smooth, spams a lot of ``[Stream 70] Invalid magic ####``, the IR projector stays on after everything closes and I have to run freenect-micview each time I plug in the Kinect to upload the firmware. (Kinect for Windows).
I think I will try to make the official primesense python library work on ARM raspi.

XHao1997, 2025-03-23:
Hi @Looki2000 , I met the same problem when installing in ubuntu20.04. Could you explain why you copy to python3.9? Is it the default version of your pi?

aidan-octane, 2025-04-03:
> I managed to make it work by copying `freenect.so` into every possible site-packages and dist-packages locations of python3/3.9 and it started to work. Sadly, it's not very smooth, spams a lot of [...]

I'm also trying to get this to work on a raspi, and am facing the same issue. I would greatly appreciate hearing what you did to get this working, or if you were able to at all.

Looki2000, 2025-04-16:
It's been a while, so I don’t remember exactly how I did it. From what I recall, after building freenect, it creates a freenect.so file somewhere, but you can’t just import it in Python right away. It doesn’t actually install it. It just builds the library. So what I ended up doing was kinda brute-forcing it: I copied that .so file into the site-packages and dist-packages folders in all the Python versions I had installed. Somehow, it started importing and working.
Copying only to the version of Python you are using should do the job, but I am unsure where should you put it - site-packages, dist-packages or both.
Keep in mind that this is a software driver and that the frame rate will likely be very low, especially on Raspberry Pi. Sadly, the official Primesense Python library does not work with Kinect because of Microsoft copyright or something, idk. It only works with Primesense Carmine, Asus Xtion or some other oem version of this sensor..

hbmartin, 2025-05-14:
You can build freenect then tell cython were to find it, take a look at the steps I'm using here: https://github.com/hbmartin/libfreenect/blob/master/.github/workflows/build.yml and https://github.com/hbmartin/libfreenect/blob/master/wrappers/python/setup.py#L51